KevinOConnor/can2040 #1: CANFD Support possible?

savejeff, 2022-07-10:
Hi!,

i just found you library. up to now i used an MCP2515 controller on the RP2040 but maybe i might switch to this. 

from your experience implementing this, would you say it could be possible to support CanFD ? 
from standard can to CanFD the Baudrate of the Payload is increased and the payload can be up to 64 bytes long. 
is that to much for the PIO ?

thanks for the Great Work ;]

KevinOConnor, 2022-07-10:
Alas, I don't have access to the canfd specs, so I don't know what it would take.

My high-level understanding is that canfd has some patent issues and that there are two competing specifications (an iso one and a bosch one).  My knowledge is pretty limited here though.

As for speed, it's technically possible to run the current can2040 code at higher data rates (up to 4Mbit/s).  However, increasing the speed would make it difficult to meet the irq latency requirements ( https://github.com/KevinOConnor/can2040/blob/master/docs/Features.md#software-utilization ).

Cheers,
-Kevin

savejeff, 2022-07-11:
Thanks for the fast response.

I've not looked into the competing specifications or patent issues. But from my knowledge of the Car industry this seems quit likely x]

But i think its not that complicated. according to [this](https://www.microcontrol.net/service/grundlagen/can-fd/)  its basically : the Baud rate increases for the data part of the msg. you then have two rates: the CAN base rate (for example 1Mbit) and the CanFD rate (for example 4Mbit). So when you send a msg with 64 DLC it takes exactly as longas a 8 DLC conventional msg. 

About the interrupt latency and processing requirements, maybe this can be solved by implementing a msg id filter on the PIO level to reduce the msgs that are forwarded to the actual RP2040 Core

KevinOConnor, 2022-08-29:
FYI, I got pointed to a CAN-FD spec from Bosch at: https://can-newsletter.org/assets/files/ttmedia/raw/e5740b7b5781b8960f55efcc2b93edf8.pdf .

I looked briefly through the CAN-FD spec.  My impression is that supporting "extended lengths" (packets up to 64 bytes) in can2040 would be reasonable to implement.  It seems that it may only need C code changes (and may not require PIO changes).

In contrast, it seems "bitrate switching" would be difficult to implement in can2040.  It seems that bitrate switching is optional though.

That said, I do not have any plans to implement either of the above.  The patent concerns limit the appeal.

Cheers,
-Kevin

savejeff, 2022-08-29:
Thanks for the feedback!

Without bitrate switching, CanFD has no advantages. 

no problem. i just wanted to know if it would theoretically be possible and get some insight. The best would would in general be to just use a CanFD chip over spi.

closed 2022-08-29 by savejeff

dc42, 2023-03-11:
CAN-FD has a significant advantage over CAN even without BRS because of the longer packet length. This means that messages longer than 8 bytes (which is practice is almost all messages) can be sent more efficiently. When using plain CAN these messages need to be chopped into fragments of ~7 bytes each, with each fragment suffering the overheads of the preamble and postamble, and the MCU has to do the extra work of fragmenting and de-fragmenting the packets. With CAN-FD a single message can be up to 64 bytes long.

We have an implementation of a subset of CAN-FD running on the RP2040, using the same PIO code that Kevin wrote for this project.  It doesn't support BRS and it doesn't support plain CAN packets. The code is in https://github.com/Duet3D/CoreN2G/tree/3.5-dev/src/RP2040. Note, CAN-FD is protected by Bosch patents.