KevinOConnor/can2040 #2: Wrong logic for TX vs RX at EOF

kentindell, 2022-07-10:
The CAN spec requires that the transmission of a message is complete at the end of the last bit of EOF, but that the reception of a message is complete at the end of the second-to-last bit of EOF (see 11989 section 10.7). It has to be this way because of CAN's atomicity property. There is a consequential feature of the protocol of the Double Receive Problem. See here for more on that: https://kentindell.github.io/2020/07/11/can-atomic-multicast/

KevinOConnor, 2022-07-10:
Can you provide some additional information on why you believe this is not working correctly?  I did attempt to make that work (see the check in `data_state_update_eof1()` in can2040.c).  Admittedly, it is a little difficult to test that particular code path.

-Kevin

kentindell, 2022-07-10:
In terms of the logic, I see you're not throwing errors and are post-processing the received signal. That corner cutting kind of works elsewhere in the frame but it doesn't apply here because of the way CAN's atomicity property works: signalling for a retransmit must happen in exactly the right place to bring all the controller into sync and for a retransmit to happen.

KevinOConnor, 2022-07-10:
> I can help you test that code path with the CANHack toolkit, which is designed to inject these conditions:

Thanks.

> In terms of the logic, I see you're not throwing errors 

The can2040 code never generates error frames.  This is a known limitation of the project.  https://github.com/KevinOConnor/can2040/blob/master/docs/Features.md#protocol-details

In regard to this issue report, it would help to have further details.  Can you confirm that you are running can2040 locally and are able to reproduce a case where can2040 does not handle the EOF bits correctly?  What frame is not being processed correctly?  What is the incorrect behaviour (for example, does can2040 internally pass on a received message that it should not have, does  can2040 fail to retransmit a message that it should have)?

Cheers,
-Kevin

kentindell, 2022-07-10:
This atomicity issue here is actually a consequence of not generating errors: the reason for the one bit time difference between TX and RX is there precisely to achieve atomicity. You should probably look again at that error restriction because the error handling in CAN is an intrinsic part of a lot of the properties of CAN: it's not just about the receiver discarding a bad frame like with Ethernet. For example, all the logic for the ACK bit is dependent on there being error signalling (and the logic for TEC and error passive is built the way it is because of the ACK field).

closed 2022-11-28 by github-actions[bot]