KevinOConnor/can2040 #4: Protability

heezes, 2022-07-15:
Hi,
First of all big thanks for writing software implementation of can. This is something i hope existed earlier but nonetheless cheers! I just wanted to know if are you planning to make microcontroller agnostic. I myself is tempted to port this version to use it in one of my project.

KevinOConnor, 2022-07-16:
I don't have any plans to port the code to other micro-controllers.

For what it is worth, the current code is tightly integrated with the rp2040 PIO hardware, which is only available on the rp2040 micro-controller.  I suspect it would be difficult to integrate with other micro-controllers.

Cheers,
-Kevin

dpharris, 2022-08-02:
Kevin --
Thank you for this well-written library.  Kudos.  
Can't wait to try it out with OpenLCB/LCC - a model rail Local Control Bus.  
Cheers,
David

closed 2022-11-28 by github-actions[bot]