KevinOConnor/can2040 #5: Licensing

coderkalyan, 2022-07-19:
Hello! 

First of all, thank you so much for writing this project! I've been hoping/thinking about this for a while but never had the expertise to make it happen. This allows me to add CAN to a lot of my projects since STM32 lineups are out of stock.

I respect and understand your decision to keep this copyleft via GPL. Unfortunately, this disallows me from encorporating your code into new RP2040 CAN drivers I would like to write for the NuttX and Zephyr RTOS's, which are both Apache 2.0 licensed. My relevant downstream projects with these are open source (I eventually hope to encorporate it into the open source PX4 autopilot which runs on NuttX) but cannot do so currently. Would you be willing to license this under a more flexible alternative?

If not, I understand and will probably maintain an out of tree driver, but I would prefer not to.

Thanks!

KevinOConnor, 2022-07-21:
I understand.  In general, I am open to custom licensing on an individual case-by-case basis.  However, I am not planning to re-license can2040 globally at this time.

Cheers,
-Kevin

closed 2022-11-28 by github-actions[bot]

thestumbler, 2023-04-09:
Just curious, have you thought about LGPL3? It isn't clear now whether a potential project I'd like to use this on, whether the client will require the application itself (not the supporting libraries) to be closed source or not. Just wondering if LGPL was something you specifically rejected or just haven't considered.
-Chris

dsseng, 2025-05-25:
@KevinOConnor Hello!

Could you please let me know if you would allow an Apache-2.0 licensing occasion for me to integrate the driver into Zephyr RTOS?

I would really like to use RP2040 in a project needing Zephyr, and seek for a way to integrate. Of course the files would retain your authorship data, however for incorporating in Zephyr repo the code (which will be a my derivative of your work) must be compatible with Apache-2.0.

It seems beneficiary for all members of the community, but I can also understand that if you are not willing to distribute your work under a non-copyleft license. I could think about using your work as a more separate module (retaining GPL), however that would come with less flexibility and quality of integration into our project

Thank you!

KevinOConnor, 2025-05-26:
I am not planning to globally re-license can2040 at this time.

Cheers,
-Kevin

KevinOConnor, 2025-05-28:
If you want to contact me directly about licensing you can reach me via email. My email is at the top of the can2040 C code.

Cheers,
-Kevin