KevinOConnor/can2040 #9: PIO1 not working

knight37x, 2022-09-12:
Hi,
thanks for this great implementation!
I am, however, struggling to get PIO1 to work (minimal example code see down below). PIO0 works fine but as soon as I set pio_num to 1 (2nd argument in can2040_setup call) and change irq nums from PIO0_IRQ_0 to PIO1_IRQ_0 (in all three lines after "Enable irqs" comment) its not working. I checked using an Oszilloscope: I don't even see any initial signal edges (e.g. of first bit).
Do you see any problem in my code? Is it a problem of this project? Is it a usage error?

Cheers,
Adrian

```
#include <stdio.h>
#include <intctrl.h>
#include "pico/stdlib.h"
#include "can2040.h"

typedef struct can2040 CANHandle;
typedef struct can2040_msg CANMsg;

static CANHandle can1;

static void can2040_cb1(CANHandle *cd, uint32_t notify, CANMsg *msg) {
  printf("can1 callback called. id: &d", notify);
}

static void PIO1_IRQHandler(void) {
  can2040_pio_irq_handler(&can1);
}

void canbus_setup() {
  uint32_t sys_clock = 125000000, bitrate = 250000;

  // Setup canbus
  can2040_setup(&can1, 1);
  can2040_callback_config(&can1, can2040_cb1);

  // Enable irqs
  irq_set_exclusive_handler(PIO1_IRQ_0, PIO1_IRQHandler);
  irq_set_priority(PIO1_IRQ_0, 1);
  irq_set_enabled(PIO1_IRQ_0, true);

  // Start canbus
  can2040_start(&can1, sys_clock, bitrate, 8, 10);
}

int main() {
  stdio_init_all();

  sleep_ms(10000);
  printf("Startup delay over\n");

  printf("Starting Initialization of CAN\n");
  canbus_setup();
  printf("Initialized CAN1\n");

  sleep_ms(1000);
  while (true) {
    CANMsg msg = {0};
    msg.dlc = 8;
    msg.id = 34;
    msg.data[0] = 11; msg.data[1] = 22; msg.data[2] = 33; msg.data[3] = 44; msg.data[4] = 55; msg.data[5] = 66; msg.data[6] = 77; msg.data[7] = 88;
    int res = can2040_transmit(&can1, &msg);
    printf("Sending! returned: %d\n", res);
    sleep_ms(1000);
  }
}
```

KevinOConnor, 2022-09-12:
I don't have my test setup immediately available, but does the following change to the can2040.c code resolve the issue?
```patch
--- a/src/can2040.c
+++ b/src/can2040.c
@@ -368,8 +368,6 @@ pio_sm_setup(struct can2040 *cd)
     pio_hw->ctrl = 0x07 << PIO_CTRL_SM_ENABLE_LSB;
 }
 
-#define PIO_FUNC 6
-
 // Initial setup of gpio pins and PIO state machines
 static void
 pio_setup(struct can2040 *cd, uint32_t sys_clock, uint32_t bitrate)
@@ -389,8 +387,9 @@ pio_setup(struct can2040 *cd, uint32_t sys_clock, uint32_t bitrate)
     pio_sm_setup(cd);
 
     // Map Rx/Tx gpios
-    rp2040_gpio_peripheral(cd->gpio_rx, PIO_FUNC, 1);
-    rp2040_gpio_peripheral(cd->gpio_tx, PIO_FUNC, 0);
+    uint32_t pio_func = cd->pio_num ? 7 : 6;
+    rp2040_gpio_peripheral(cd->gpio_rx, pio_func, 1);
+    rp2040_gpio_peripheral(cd->gpio_tx, pio_func, 0);
 }
 
 
```

-Kevin

knight37x, 2022-09-12:
I will try next two days and report back

knight37x, 2022-09-14:
Yeah! That makes the trick! Thanks for the quick solution!

KevinOConnor, 2022-09-14:
Thanks.  I committed the corresponding change (commit e60fbf65).

-Kevin

closed 2022-09-14 by knight37x