KevinOConnor/can2040 #13: PIO not running?

zbrown23, 2022-11-06:
Hi All,

I'm really looking forward to getting this library working, but I've been fighting with it a decent amount as time goes on. 

currently my code looks as follows:

```c
#include "can2040.h"
#include "pico/stdlib.h"
#include "hardware/regs/intctrl.h"
#include <stdio.h>

struct can2040 can;

static void can2040_cb1(struct can2040 *cd, uint32_t notify, struct can2040_msg *msg) {
    printf("can callback called. id: &d", notify);
}

static void PIO0_IRQHandler(void) {
    can2040_pio_irq_handler(&can);
}

void canbus_setup() {
    uint32_t sys_clock = 125000000, bitrate = 250000;

    // Setup canbus
    can2040_setup(&can, 0);
    can2040_callback_config(&can, can2040_cb1);

    // Enable irqs
    irq_set_exclusive_handler(PIO1_IRQ_0, PIO0_IRQHandler);
    irq_set_priority(PIO1_IRQ_0, 1);
    irq_set_enabled(PIO1_IRQ_0, true);

    // Start canbus
    can2040_start(&can, sys_clock, bitrate, 0, 1);
}

int main() {
    stdio_init_all();
    canbus_setup();
    printf("Initialized CAN\n");
    sleep_ms(1000);
    while (true) {
        struct can2040_msg msg = {0};
        msg.dlc = 8;
        msg.id = 34;
        msg.data[0] = 0xff;
        msg.data[3] = 0xee;
        msg.data[5] = 0x11;
        int res = can2040_transmit(&can, &msg);
        printf("transmit returned: %d\n", res);
        sleep_ms(1000);
    }
}
```

When I run it, I get nothing out of pins 0 and 1, and  get a -1 from `can2040_transmit()`. This leads me to believe that the PIO FIFO is filling up and it's never starting in the first place. I do get a single pin pulled high, (either 0 or 1), but I'm not sure why that would be. 

for other reference, here's the CMakelists.txt for the target itself:
```
add_executable(can2040test
        src/main.c
        )

target_link_libraries(can2040test PUBLIC
        pico_stdlib
        hardware_i2c
        can2040
        )

pico_add_extra_outputs(can2040test) # output the uf2 file for flashing
pico_enable_stdio_usb(can2040test 1) # enable stdio over usb instead of UART
```

and the CMakelists.txt for my generic libraries folder

```
# can2040
add_library(can2040 can2040/src/can2040.c)
target_include_directories(can2040 PUBLIC
        can2040/src/
        can2040/pio/
        ../sdk/pico-sdk/src/rp2_common/cmsis/stub/CMSIS/Device/RaspberryPi/RP2040/Include/
        ../sdk/pico-sdk/src/rp2_common/cmsis/stub/CMSIS/Core/Include/
        )

target_link_libraries(can2040 PUBLIC
        cmsis_core
        hardware_irq
        pico_stdlib
        hardware_pio
        )
```

Thanks in advance!
Zach

zbrown23, 2022-11-08:
Hi Kevin,

I followed your advice and just set up my a bus with a CAN to usb dongle and a TI SN65HVD230D CAN transceiver connected to the pico. I didn't get anything out of the CAN to usb dongle, and also still got -1 from `can2040_transmit()`, although I haven't put a logic analyzer on the TX or RX lines, nor the CANH/CANL lines. I'll see what those look like and report back.

zbrown23, 2022-11-27:
Resolved by PRs associated with issue #14

closed 2022-11-27 by zbrown23

KevinOConnor, 2022-11-27:
Thanks for the update.  Glad the issue was resolved.

-Kevin