KevinOConnor/can2040 #19: Add cmake import

HondaRulez, 2022-12-09:
Hi,
pls add this file to the project to allow importing as a library (like RTOS, and so on..) into the project.

usage: add few lines to the main CMakeLists.txt of the project as follows

#Add the next line after pico_sd_init ()
include(path_to_the/can2040-import.cmake)

# edit the existing line
target_link_libraries ( your-project
  your_lib1
  your_lib2
 .
 can2040
 .
 your_libx
)

# The gzipped import cmake file:
[can2040-import.cmake.gz](https://github.com/KevinOConnor/can2040/files/10195763/can2040-import.cmake.gz)

KevinOConnor, 2022-12-12:
Thanks, but every project is likely to have notabaly different build rules.  It's not a goal of mine to track the many possible build rules in the can2040 repository.

Cheers,
-Kevin

LucaOverflow, 2023-02-08:
Hi @HondaRulez,

I really like your approach. But sadly I can't make it work in my project. Actually I'n unable to get can2040 working at all. I'm pretty sure it has to do with my cmake setup. I tried many different ways to include can2040. But no matter what I do I will always get `undefined reference to` Errors to the can2040 functions. However my IDE is happy. Would be great if you would take a look at my CMakeLists.txt below. The can2040 repository with the added .cmake file is located in the include directory. The main.cpp consists of the startup code from the API docs.

```
cmake_minimum_required(VERSION 3.25)

# initialize pico-sdk from GIT
# (note this can come from environment, CMake cache etc)
set(PICO_SDK_FETCH_FROM_GIT on)

# pico_sdk_import.cmake is a single file copied from this SDK
# note: this must happen before project()
include(pico_sdk_import.cmake)

project(my_project)

# initialize the Raspberry Pi Pico SDK
pico_sdk_init()

# can2040
include(include/can2040/can2040-import.cmake)

add_executable(${PROJECT_NAME}
    src/main.cpp
)

include_directories(
    include
)

# Add pico_stdlib library which aggregates commonly used features
target_link_libraries(${PROJECT_NAME}
    pico_stdlib
    cmsis_core
    can2040
)
```

closed 2023-04-22 by github-actions[bot]

KermMartian, 2023-06-27:
Thanks for providing this! I was trying to implement this myself, and you saved me some time trying to figure this out.