KevinOConnor/can2040 #20: Arduino library available

obdevel, 2022-12-12:
I have created a wrapper library that works with the arduino-pico Arduino core: https://github.com/obdevel/ACAN2040

It has received some limited testing and will sustain 1000 msgs/sec at 125kb/s bus speed. It has been tested with two CAN controllers ICs: MCP2562 (DIP8 package on a breadboard) and SN65HVD230 (using modules available from eBay, etc). We are currently characterising error conditions (e.g. poor wiring, connections, other nodes failing, etc).

It is currently limited to a single bus instance as that is all I need. I may add support for two concurrent instances if there is a demand for this.

I have had to include Kevin's code in my library as there doesn't seem to be a way for an Arduino library to reference external files. This means that I will need to manually track any changes that Kevin makes. One additional #include was required to successfully compile using the toolchain supplied with the arduino-pico core. You don't need to separately install the Pico SDK as this is included with the Arduino core.

Documentation is limited to an example sketch and the code itself. I'm happy to receive bug reports or PRs to fix them.

I haven't issued a formal release, created the library metadata, or registered it with the Arduino project, so it is not available from the Arduino IDE Library Manager. I may do this in the future. For now, you'll have to download and install it manually. The licence is GPL-3.0, in common with this project.

Many thanks to Kevin for this very useful project.

KevinOConnor, 2022-12-14:
Interesting.  Thanks.

I look forward to seeing the results of your tests.

Note that you should probably grab the latest version of can2040.c as there were some recent bug fixes (in particular with handling of retransmits).

-Kevin

obdevel, 2022-12-17:
I tried to integrate the new files but got some linker errors. It seems that the arduino-pico toolchain requires:
```
#ifdef  __cplusplus
extern "C" {
#endif

#ifdef  __cplusplus
}
#endif

```
in the .h file. I have manually added this for the time being.

The compiler used is:
```
Using built-in specs.
COLLECT_GCC=./Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++
COLLECT_LTO_WRAPPER=/Users/duncang/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/../libexec/gcc/arm-none-eabi/10.3.0/lto-wrapper
Target: arm-none-eabi
Configured with: /workdir/repo/gcc-gnu/configure --prefix=/workdir/arm-none-eabi.osx --build=x86_64-linux-gnu --host=x86_64-apple-darwin14 --target=arm-none-eabi --disable-shared --with-newlib --enable-threads=no --disable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-nls --without-python --disable-bootstrap --enable-languages=c,c++ --disable-lto --enable-static=yes --disable-libstdcxx-verbose --disable-decimal-float --with-cpu=cortex-m0plus --with-no-thumb-interwork
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.3.0 (GCC)
```

It also gives some compilation warnings, probably because the flags are less permissive. I doubt they're problematic but some people will complain if the build isn't clean !!

```
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c: In function 'pio_tx_send':
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:271:16: warning: comparison of integer expressions of different signedness: 'int' and 'uint32_t' {aka 'long unsigned int'} [-Wsign-compare]
  271 |     for (i=0; i<count; i++)
      |                ^
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c: In function 'pio_sm_setup':
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:355:16: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
  355 |     for (i=0; i<ARRAY_SIZE(can2040_program_instructions); i++)
      |                ^
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c: In function 'data_state_update_discard':
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:1101:56: warning: unused parameter 'data' [-Wunused-parameter]
 1101 | data_state_update_discard(struct can2040 *cd, uint32_t data)
      |                                               ~~~~~~~~~^~~~
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c: In function 'can2040_transmit':
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:1245:16: warning: comparison of integer expressions of different signedness: 'int' and 'uint32_t' {aka 'long unsigned int'} [-Wsign-compare]
 1245 |     for (i=0; i<data_len; i++) {
      |                ^
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c: In function 'can2040_shutdown':
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:1298:34: warning: unused parameter 'cd' [-Wunused-parameter]
 1298 | can2040_shutdown(struct can2040 *cd)
      |                  ~~~~~~~~~~~~~~~~^~
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c: In function 'crc_bytes':
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:439:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  439 |     default: crc = crc_byte(crc, data >> 24);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:440:5: note: here
  440 |     case 3:  crc = crc_byte(crc, data >> 16);
      |     ^~~~
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:440:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  440 |     case 3:  crc = crc_byte(crc, data >> 16);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:441:5: note: here
  441 |     case 2:  crc = crc_byte(crc, data >> 8);
      |     ^~~~
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:441:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  441 |     case 2:  crc = crc_byte(crc, data >> 8);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/<redacted>/Documents/Arduino/libraries/ACAN2040/src/can2040.c:442:5: note: here
  442 |     case 1:  crc = crc_byte(crc, data);
      |   
```

I'll write a separate message once we've completed our testing with the new code.

KevinOConnor, 2022-12-21:
Nothing has changed recently with respect to the above messages.

I did make some minor code changes to reduce some warnings.  In general though, these warnings are low quality and in my opinion shouldn't be enabled in a build.  You should be able to alter the gcc command-line parameters to suppress the remaining (not useful) warnings.

If you are importing a C header file into a C++ file you can wrap the import in the including file with something like:
```c
extern "C" {
#include "can2040.h"
}
```

-Kevin

closed 2023-02-28 by github-actions[bot]