KevinOConnor/can2040 #25: Documentation / Application Reference

6ffm70, 2023-01-19:
can2040 looks like it could help me with a project that reads data from my motorcycle with an Arduino Nano Connect. Is there a description of a reference application with a single-board microcontroller somewhere?

Thanks

KevinOConnor, 2023-01-25:
There is no description of a reference hardware.  One can use Klipper as a reference software implementation: https://github.com/KevinOConnor/can2040/blob/master/docs/Tools.md 

-Kevin

thestumbler, 2023-03-24:
Could you remark which CAN transceiver chip you used in your experiments? Are there any reasons to prefer one particular transceiver over any others?

KevinOConnor, 2023-03-27:
There is some info on test hardware I've used at https://github.com/KevinOConnor/can2040/blob/master/docs/Tools.md#testing-with-raspberry-pi-pico-board .  I don't have a lot of experience with different canbus transceivers - as far as I know, as long as a transceiver supports 3.3V digital IO, it should work with can2040.

-Kevin

thestumbler, 2023-03-27:
Thanks Kevin. Just one day after I posted this question, I dug around more in your repository and found the information you mention that I'd missed on my first search. 

I'm gonna just try with a couple of AliExpress Can transceivers that I have on-hand from another project and see what happens. I have one of the RP2040 Pico - Can modules, but it has the transceiver and the Can interface chip. As convenient as it would be, I don't think I could use that board without butchering it irreversibly (I really want to test with your technique to avoid being dependent on a transceiver chip which may be hard to source). Plus I want to get some experience with the PIOs

closed 2023-06-04 by github-actions[bot]