KevinOConnor/can2040 #28: can2040 for micropython

Eminemchik, 2023-02-27:
Hi, thanks for the great library!
Are there plans for a micropython version?

bjpirt, 2023-02-28:
I came here wondering the same thing - would be an amazing add-on to the MicroPython ecosystem!

KevinOConnor, 2023-03-07:
FWIW, I don't know of anyone currently working on a MicroPython implementation.

Cheers,
-Kevin

bjpirt, 2023-03-11:
@KevinOConnor Given that you can configure the RP2040 state machines directly from Python code, I was wondering if you know of any blockers to porting your code to pure Python (given you know the code best!), rather than compiling the C code into MicroPython.

I wouldn't mind giving it a try but knowing it's theoretically possible would be helpful before embarking on such a mission

KevinOConnor, 2023-03-11:
Alas, it's not something I've looked at, so I'm not able to provide any meaningful advice.

-Kevin

ferranmercader, 2023-05-05:
@Eminemchik @bjpirt , I'm came here also wondering the same thing. I've found this repository:
https://github.com/kentindell/canis-can-sdk

Did you finally find any solution?

Eminemchik, 2023-05-15:
@ferranmercader I ended up using the MCP2515 in my project. I used the code from https://www.waveshare.com/wiki/Pico-CAN-B as an example.

closed 2023-07-23 by github-actions[bot]