KevinOConnor/can2040 #35: Recovery when an error occurs

furusawata, 2023-04-25:
Hello Kevin

I am writing a program using can2040.
An error is intentionally generated at the time of transmission and reception, and the behavior is confirmed.
If an error occurs during transmission, recovery is not possible.

I also tested it with Klipper.
`cangen can0 -g 2 -e -I 00EF0102 -L 8 -D 0000000000000000`
It stops when an error is generated on purpose.

[image]
The command stops and prints the following comment:
`write: No buffer space available`
After this I can't restore without unplugging and plugging the USB.

Can I restart with can2040 software?

KevinOConnor, 2023-04-25:
I'm not sure what you are reporting.  If there is a situation where can2040 is not doing the right thing, I'd need to be able to setup a test environment here to reproduce the problem.  Can you describe your test hardware and the steps needed to reproduce the problem?

-Kevin

furusawata, 2023-04-26:
The following configurations have been tested.

[image]

Packets are sent from A to C or B to C.

The following is the waveform under normal conditions.

[image]

PCAN-USB can error on any bit.

[image]

When an error is made from B to C, it is resent.

[image]

However, it does not resend when an error is generated from A to C.

[image]

Is there a problem with my can2040 usage?
Is there a way for you to raise the error?
Is there anything else to check?

KevinOConnor, 2023-04-26:
Okay I think I understand.  You are reporting that if another node sends an error frame while can2040 is transmitting then can2040 stops functioning.

I don't have a hardware setup readily available to reproduce this.  However, I've seen lots of cases where can2040 correctly responds to an error frame.  So, this sounds like a regression or something specific to the timing of the error frame.

What version of can2040 are you using?

Have you tried running an older version of can2040 to see if it has the same behavior (for example the v1.3.0 tag or v0.2.0 tag of can2040)?

Does injecting the error frame at a different bit location (eg, bit 20 instead of 60) alter the outcome of the test?

I'll see if I can setup a test environment locally.

-Kevin

furusawata, 2023-04-26:
I tried changing the bit location.
Bit101 and below were not resent, and bit102 and above were resent.

[image]

There was no difference between versions.

- master
- work-txtiming-20230313
- can2040-1.3.0
- can2040-0.2.0

Regards,

KevinOConnor, 2023-04-26:
You have very nice test equipment.

I was able to reproduce this problem locally.  It appears the can2040 code has a defect in how it detects "tx conflict" conditions.

I have a proposed fix at #36.

Thank you for reporting this issue.
-Kevin

furusawata, 2023-04-27:
Thanks for the fix.
Confirmed to resend.

Could you also fix work-txtiming-20230313(#32)?

KevinOConnor, 2023-04-27:
Great.

Sure, I rebased #32 on top of #36.

-Kevin

KevinOConnor, 2023-04-29:
I committed PR #36.

-Kevin

closed 2023-05-21 by KevinOConnor