KevinOConnor/can2040 #37: Add extern "C" {} block for C++ compatibility

Nate711, 2023-05-03:
I was having linking errors using this library until I realized there was no extern "C" block included. I think it would be a nice feature thanks!

KevinOConnor, 2023-05-03:
If you are importing a C header file into a C++ file you can wrap the import in the including file with something like:

```c
extern "C" {
#include "can2040.h"
}
```

-Kevin

Nate711, 2023-05-03:
Thanks, it's just that I spent a long time trying to figure out the linking error when it was just the extern C issue. If the library comes with the extern C blocks built in to the can2040.h file, then newbies like myself won't have to repeat the same struggle. Thanks!

chuggafan, 2023-05-11:
Would it not be possible to just add:
```cpp
#ifdef __cplusplus
extern "C" {
#endif
// header file
#ifdef __cplusplus
}
#endif
```
to the header file as well? This would solve the underlying issue and should not overall pollute any of the code.

KevinOConnor, 2023-05-21:
I'm aware that one could use an `#ifdef`, but would prefer to keep the headers focused on C code for now.

Closing as this should be addressed with the documentation update.

-Kevin

closed 2023-05-21 by KevinOConnor