KevinOConnor/can2040 #40: Tx retry count.

HondaRulez, 2023-05-20:
Hi, 
Any plan to add a "max retry count" when transmitting ? (like on a empty BUS)
anding a new notify like CAN2040_NOTIFY_TX_CANCELLED when the given packet was cancelled.

KevinOConnor, 2023-05-21:
I don't have a plan to implement that.  I don't know of anyone working on it.

Cheers,
-Kevin

linted, 2023-06-08:
+1 
This would be an extremely useful feature. I have a use case where the other nodes on a canbus may not be present, and being able to stop automatic retransitions would to reset internal state would be extremely useful.

HondaRulez, 2023-06-08:
I will do this mod at the weekend because getting very important to me :), I will push here the mods as a pull request...

linted, 2023-06-30:
@HondaRulez please take a look at #40 to see if it works for your use case as it does mine.

closed 2023-09-07 by github-actions[bot]