KevinOConnor/can2040 #43: Pico stops sending can message halfway through, then retries

Nate711, 2023-06-13:
Hi, I was having no problems when talking between two can transceivers connected to the same pico, but now when I talk to an external device (mg4005 can bus bldc motor), the Pico seems to try to send a message, fails halfway, then retries successfully. Any idea what's going on? Thanks!

[image]

KevinOConnor, 2023-06-16:
That certainly looks odd.  Make sure you are on the very latest can2040 code as there were some recent changes to the handling of transmit messages.

-Kevin

Nate711, 2023-06-17:
Thanks for the response Kevin! I just checked and I'm using the latest main. It seems like I'm able to change how often these errors occur by changing the delay between sending messages. For instance, if I delay 200us between sending two CAN messages, the error rate seems to be low, like 1 in 10. However if I change that to 300us, it becomes maybe 1 in 2. Btw, the code I'm running is this: https://github.com/Nate711/can2040-test/blob/main/apps/mg4005_test.cc

Nate711, 2023-06-17:
I resolved one issue: In addition to the half-formed can messages, the pico would often (~1 in 3 times) repeat even a successfully sent and ACKd message (see below). I fixed this by just removing my print statement that happened every time CAN messages were sent. Oops! I didn't realize the USB comms would add irq latency. Still have fix the half-formed message problem...
[image]

Nate711, 2023-06-17:
Update: I removed the actuator and instead made CAN 1 talk to CAN 2 (both have transceivers) to see if the actuator was the issue. It was not. The program I used to test this sent 3 messages over can in a 1khz loop. Sometimes when I run my code, the 2nd message is repeated, but other times, the messages are totally fine. Maybe you can recreate this issue on your side. My code is: https://github.com/Nate711/can2040-test/blob/main/apps/dual_can_test.cc 
[image]

closed 2023-08-27 by github-actions[bot]