KevinOConnor/can2040 #48: undefined reference to `__DMB'

danielkucera, 2023-10-18:
I am getting following error when compiling the library:
```
Compiling .pio/build/rp2040/libdb9/can2040/can2040.c.o
.pio/libdeps/rp2040/can2040/src/can2040.c: In function 'tx_schedule_transmit':
.pio/libdeps/rp2040/can2040/src/can2040.c:666:9: warning: implicit declaration of function '__DMB' [-Wimplicit-function-declaration]
  666 |         __DMB();
      |         ^~~~~
Archiving .pio/build/rp2040/libdb9/libcan2040.a
Indexing .pio/build/rp2040/libdb9/libcan2040.a
Linking .pio/build/rp2040/firmware.elf
/home/danman/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/rp2040/libdb9/libcan2040.a(can2040.c.o): in function `tx_schedule_transmit':
can2040.c:(.text.tx_schedule_transmit+0x4e): undefined reference to `__DMB'
collect2: error: ld returned 1 exit status
*** [.pio/build/rp2040/firmware.elf] Error 1

```
I managed to get rid of the error by following patch:
```
diff --git a/src/can2040.c b/src/can2040.c
index c2bd006..8b4abd5 100644
--- a/src/can2040.c
+++ b/src/can2040.c
@@ -14,7 +14,7 @@
 #include "hardware/structs/padsbank0.h" // padsbank0_hw
 #include "hardware/structs/pio.h" // pio0_hw
 #include "hardware/structs/resets.h" // RESETS_RESET_PIO0_BITS
-
+#include "cmsis_gcc.h"
 
 /****************************************************************
  * rp2040 and low-level helper functions
```
Is this the correct way? Is the include really missing?

KevinOConnor, 2023-10-19:
I'm not sure why that would be.  That header should be already included via `RP2040.h -> cmsis_cm0plus.h -> cmsis_compiler.h -> cmsis_gcc.h` .  So, the header you've found is the correct location of the definition, but it should have already been included.

Cheers,
-Kevin

danielkucera, 2023-10-19:
Oh, right, I also commented RP2040.h, with that included I was getting another error:
```
Building in release mode
Compiling .pio/build/rp2040/libdb9/can2040/can2040.c.o
.pio/libdeps/rp2040/can2040/src/can2040.c:9:10: fatal error: RP2040.h: No such file or directory

****************************************************************
* Looking for RP2040.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:RP2040.h"
* Web  > https://registry.platformio.org/search?q=header:RP2040.h
*
****************************************************************

    9 | #include "RP2040.h" // hw_set_bits
      |          ^~~~~~~~~~
compilation terminated.
*** [.pio/build/rp2040/libdb9/can2040/can2040.c.o] Error 1
```

```
diff --git a/src/can2040.c b/src/can2040.c
index c2bd006..ad8abcb 100644
--- a/src/can2040.c
+++ b/src/can2040.c
@@ -6,7 +6,7 @@
 
 #include <stdint.h> // uint32_t
 #include <string.h> // memset
-#include "RP2040.h" // hw_set_bits
+//#include "RP2040.h" // hw_set_bits
 #include "can2040.h" // can2040_setup
 #include "hardware/regs/dreq.h" // DREQ_PIO0_RX1
 #include "hardware/structs/dma.h" // dma_hw
@@ -14,7 +14,7 @@
 #include "hardware/structs/padsbank0.h" // padsbank0_hw
 #include "hardware/structs/pio.h" // pio0_hw
 #include "hardware/structs/resets.h" // RESETS_RESET_PIO0_BITS
-
+#include "cmsis_gcc.h"
 
 /****************************************************************
  * rp2040 and low-level helper functions
  * 
```

I am using Platform.io with Arduino for rpi-pico, here is the complete project:
https://github.com/danielkucera/rpi-can-relay

closed 2023-12-27 by github-actions[bot]

matonga, 2024-12-05:
Hi, I'm having the same problem.

I could resolve the RP2040.h issue after editing `CMakeLists.txt` and adding `cmsis_core` to `target_link_libraries`.

But it's still giving undefined reference to __DMB to me too.

matonga, 2024-12-05:
Ok, me bad! I had commented RP2040.h line in can2040.h :blush:

So:

* Ensure RP2040.h line is uncommented in can2040.h
* Add cmsis_core to target_link_libraries in CMakeLists.txt
* Rerun cmake in your build dir