KevinOConnor/can2040 #60: Only able to receive and send 4 messages then locks up? Pointers pls?

Rich235, 2024-11-14:
Hello Kevin, thank you for developing this library. 

I've a new RP2040-Zero connected to SN65HVD230 and a PCAN-USB interface with termination. I'm using very simple sketch to receive and send. I've tried 50k and 500k bps and different message sending rates, but **after 4 messages sent, it then fails to receive or send**. 

I'm using pins 9 and 10. May I ask if you have any pointers to help me debug this please? 

I'm using the latest Pico SDK with PlatformIO and Earlephilhower platform. Any guidance much appreciated. I'm new to using PIO. Thanks very much, Richard.

Rich235, 2024-11-15:
Just to add to this, I've found I can just receive fine, continually, but if I send more than 4 messages, all CAN comms fail from then on. If you can point me to where I should start looking, that would be very helpful. This is with SDK 2.0.0. Thank you.

obdevel, 2024-11-23:
That is the size of the transmit buffer and indicates the messages are not being destaged to the bus. Check your wiring, termination, etc. You should also call can2040_check_transmit() before transmitting to check there is sufficient space in the transmit buffer. If there isn't, you can wait, discard or buffer the message yourself.

If you're using Earle's Arduino core, you may find it easier to use my Arduino library which wraps Kevin's code. It includes an example sketch. Look for ACAN2040 in the IDE library manager. I've just updated it to capture Kevin's latest changes and to correctly support the RP2350 chip.

Rich235, 2024-11-23:
Thanks yery much, that is very likely it and I've not checked that yet. Yes, I've tried your (obdevel) ACAN2040 library and https://github.com/kaminaris/can2040.git, all with the same issue so far. May I ask what size of message are you able to send and at what period on a 500kbs bus? I will ensure to use can2040_check_transmit() before transmitting.

obdevel, 2024-11-24:
We have tested my Arduino library extensively but our standard rate is 125kbps. Our application does rate-limit transmissions to allow other nodes time to speak. 125kbps is around 1000-2000 messages per sec, depending on message length. We also buffer incoming and outgoing messages. If can2040_check_transmit() fails, we buffer in our own circular buffer and try again a little later. Unfortunately, Kevin advises against transmitting a message from with the callback function, so we set a flag and transmit from the main loop.

closed 2025-02-01 by github-actions[bot]