KevinOConnor/can2040 #67: Monitor mode

matonga, 2025-02-05:
I'd love to see a "monitor" mode. This would not send ACKs or any other stuff, it would only receive canbus frames and just discard invalid frames without ever writing to the tx pin.

I understand could just not connect the tx pin to the canbus transceiver, but would prefer to have the ability to switch between normal mode and monitor mode in software. Tried to implement this myself but the PIO state machine seems to depend on successful transmission of the ack to prepare for reception of next frame and couldn't solve it.

kaseiiro, 2025-02-09:
A simple (software!) workaround is to remap the tx to an unused pin (something like gpio_tx = 28 instead of 5; maybe -1 works too). Please give it a try.

KevinOConnor, 2025-02-16:
Interesting.  It should be possible to implement by changing the C code.  It's not something I've looked at (and I'm not sure I'll get a chance to look at it in the near term).

As indicated above, designating a dummy tx pin may work.

Cheers,
-Kevin

closed 2025-04-25 by github-actions[bot]