KevinOConnor/can2040 #68: Unstable / unusable

brandonros, 2025-03-01:
Preamble:

* I'm using https://github.com/KevinOConnor/can2040/pull/66 and I need to try just using the latest master instead
* I'm using "Rust bindings" and not raw C: https://github.com/brandonros/can2040_rs
* I'm using with cyw43 + Rust: https://github.com/brandonros/rp2350-isotp-ble-bridge

I get a slew of CAN errors, failed TX (first few messages work and then transmit says the tx queue is full even though nothing on the CAN network changed), failed RX, etc.

Any tips on how to debug/provide logs to show what the problem is? Like... is my rp2350 bad? I tried 2 CAN transceivers, different wires...

Is this expected to be proof-of-concept or like "good to go"?

closed 2025-03-09 by brandonros

Alia5, 2025-03-30:
@brandonros 

Since you closes the issue, I assume you've found a solution?
Please don't make this another [DenverCoder](https://xkcd.com/979/) situation...

brandonros, 2025-03-30:
I didn't but I didn't think it was right to waste the resources of contributors to help me debug unless they were interested

KevinOConnor, 2025-03-30:
The can2040 code is widely used.  If you're having issues, try following the steps at https://github.com/KevinOConnor/can2040/blob/master/docs/Tools.md .  In particular, try verifying your hardware with the known working Klipper "usb to can bus bridge" setup.

Cheers,
-Kevin