KevinOConnor/can2040 #71: Roadblock with rp2350

FutureJames, 2025-04-17:
Working on a very basic proof of concept with a Raspberry Pi Pico 2W and this can2040 project.  

I have a simple setup that works fine when I compile and run it on a Pico1 (2040).  When I take the exact same code it compiles for the rp2350, and runs but it will absolutely not send or receive any messages.  

This is all running on a test bench right now.  I pull the pico1 out of the socket and put the pico2 in.  So not a wiring or CANBUS issue.  I ran both pico1 and pico2 on the openOCD debugger and they seem to initialize exactly the same and I get no errors or unusual behavior from the pico2.  There are no runtime exceptions either, it just doesn't send or receive.  Any ideas would be appreciated.

My single file is this main.c:

```

#include "pico/stdlib.h"
#include <stdbool.h>

//Minimal includes for CAN2040 
#include <stdio.h>
#include "can2040.h"

// Determine if the target is an rp2350
#ifdef PICO_RP2350
  #include "RP2350.h"
#else
  #include "RP2040.h"
#endif

static struct can2040 cbus;

static void
can2040_cb(struct can2040 *cd, uint32_t notify, struct can2040_msg *msg)
{
    // Add message processing code here...
    printf("CAN message received: ID=0x%03X, DLC=%d, Data=", msg->id, msg->dlc);
}

static void
PIOx_IRQHandler(void)
{
    can2040_pio_irq_handler(&cbus);
}

void canbus_setup(void)
{
    uint32_t pio_num = 0;
    uint32_t sys_clock = 125000000, bitrate = 500000;
    uint32_t gpio_rx = 4, gpio_tx = 5;

    // Setup canbus
    can2040_setup(&cbus, pio_num);
    can2040_callback_config(&cbus, can2040_cb);

    // Enable irqs
    irq_set_exclusive_handler(PIO0_IRQ_0_IRQn, PIOx_IRQHandler);
    NVIC_SetPriority(PIO0_IRQ_0_IRQn, 1);
    NVIC_EnableIRQ(PIO0_IRQ_0_IRQn);

    // Start canbus
    can2040_start(&cbus, sys_clock, bitrate, gpio_rx, gpio_tx);
}

int main()
{
    stdio_init_all();
    canbus_setup();
    printf("CAN bus setup complete.\n");
} 
```

KevinOConnor, 2025-04-17:
If you're having an issue I'd recommend following the testing steps outlined at https://github.com/KevinOConnor/can2040/blob/master/docs/Tools.md - in particular verify with known working code.

For what it is worth, though, it seems unlikely that you'd be able to use the same `sys_clock` on both rp2040 and rp2350.

-Kevin

FutureJames, 2025-04-18:
okay you hit the nail on the head.  The issue was the sys_clock.  I have updated my minimal implementation project to account for this.  The code now dynamically updates the included libraries at compile time and the system clock speed at runtime.

Thanks for sharing this project Kevin.  I really don't know the first thing about writing a PIO so this work is really appreciated.  Feel free to link my [minimal implementation](https://github.com/FutureJames/can2040_pico2_minimal), or just add it to this repo as an example.  I do think that setting up Klipper might be a little overkill for testing, since it does not really demonstrate the absolute steps for someone including this library in their own project.

Cheers,
James

closed 2025-04-18 by FutureJames